TheThingsArchive/ttn #736: Device address cannot be set with the API using ABP

FlorianCodes, 2018-08-30:
@htdvisser 
Unfortunately, closed issue #733 is not solved.

> Network addresses (starting with 26 or 27) need to be assigned by the network.

This is not the case. 

In order to use ABP it is necessary to provide the device and the LNS with the device's address, before the device connects to the network.

If a device is registered in the LNS using the API and the "dev_addr" parameter is not provided in the request, then the device address is not created automatically and cannot be set manually afterwards.

johanstokking, 2018-09-03:
> In order to use ABP it is necessary to provide the device and the LNS with the device's address, before the device connects to the network.

It's the other way around; the network gives you the ABP device address. Therefore, we don't allow custom device addresses on The Things Network.

There's an API call `GetDevAddr()` to get a server generated device address to be used in ABP, see https://github.com/TheThingsNetwork/api/blob/master/protocol/lorawan/DevAddrManager.md

closed 2018-09-03 by johanstokking

Niondir, 2018-09-04:
That means use cases where I have a fixed dev address are not covered by TTN? Is there any plan in future to support custom dev addresses? Other network operators do so.

johanstokking, 2018-09-04:
It can be supported on private networks only.

Public networks that are supporting custom DevAddrs do that for legacy reasons or because they do not understand LoRaWAN.