TheThingsArchive/ttn #738: No Downlink after register the same Device while just changing the AppEUI

Niondir, 2018-09-24:
This is a **bug report** for **the backend/the console**.

> What do you want to do?
Register an already existing device in another app.

> What steps did you take?
* Register device in App A
* Test the Device (OKAY)
* Register the same Device in App B (Just Change the AppEUI, keep DevEUI and AppKey)
* Test the Device (OTAA fails)

> What went wrong or what is missing?
The OTAA join is listed in the Console but the downlink is not received by Device

Changing the AppKey as well solves the issue as a workarround. 

> What are the IDs and EUIs of your Device/Gateway? (if applicable)
DevEUI: 0004A30B00008CB0

htdvisser, 2018-09-25:
I can not reproduce this. @johanstokking did you reproduce this before assigning me to the issue?

Niondir, 2018-09-25:
I should be able to reporoduce it with the exact same device. If it helps let me know in what state I can put the device, I can have it rejoin all the time if you are able to debug it then .

I can have a look into it tomorrow.

johanstokking, 2018-09-26:
Please do that.

closed 2021-06-03 by htdvisser