TheThingsArchive/ttn #746: Changing device settings, even just its description, resets the frame counters

avbentem, 2018-11-05:
This is a **bug report** for **the console**.

> What do you want to do?

Change some settings of a device in TTN Console.

> What steps did you take?

Changed the human-readable device's description. (Same goes for any other setting, including changing "Frame Counter Checks".)

> What went wrong or what is missing?

Both the uplink and downlink frame counters were reset to zero, basically requiring the device to be reset.

(If this is by design, then a warning should be shown.)

> Do you have Screenshots?

N/A.

> What kind of OS/Browser/Gateway are you using? Which version?

N/A.

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

N/A.

> What do your configuration files look like?

N/A.

> What do your log files look like?

N/A.

> Can you fix this yourself and submit a pull request?

No.

johanstokking, 2018-11-05:
Thanks, we're adding some explanation in the V3 Console for this.

avbentem, 2018-11-06:
Hmmm, that's weird, I can really reproduce this every time:

[image]

(Sorry for the slow GIF...)

avbentem, 2018-11-06:
@htdvisser Indeed, for me it's only happening with an ABP device. Not such a common problem after all then, though a warning might be nice...

(Using `ttnctl` for the very same ABP device does not reset the counters.)

htdvisser, 2021-06-03:
This is not going to get fixed in V2 anymore.

closed 2021-06-03 by htdvisser