TheThingsArchive/ttn #749: Addition of Python Brackets instead of Braces on Console

brentru, 2018-11-28:
Hi TTN Team,

I'd like to request a feature on the web console for applications which would allow for easy inclusion of Python-nodes, in addition to C/C++ based nodes.

In Python, `{ }` creates a dict, and `[ ]` creates a list: they're both collections of objects, but they aren't the same. 

Currently the TTN Console only prints keys formatted with braces `{ }`. Would it be possible to add a button or switch switch to assign brackets `[ ]` around the keys?

brentru, 2018-11-29:
Great, thanks!

htdvisser, 2019-02-18:
I submitted https://github.com/TheThingsNetwork/lorawan-stack/pull/155 for v3 to only render the contents, and leave the braces/brackets out. Toggling through different array/list styles for different languages wouldn't improve the user experience.

closed 2019-02-18 by htdvisser