TheThingsArchive/ttn #756: TTN Integration: Issue with AWS IoT integration

Maverick10, 2019-02-24:
I'm trying to integrate AWS IoT with TTN. When trying to create the cloudformation stack I'm faced with this error
`No Solution Stack named '64bit Amazon Linux 2018.03 v2.10.0 running Go 1.11.4' found. (Service: AWSElasticBeanstalk; Status Code: 400; Error Code: InvalidParameterValue; Request ID: 27a55fac-ef00-4ee2-8c06-fcc024e23836)`

> What do you want to do?
Integrate AWS IoT with TTN

> What steps did you take?
Followed this [quick start](https://www.thethingsnetwork.org/docs/applications/aws/quick-start.html). Filled the required fields to create the cloudformation stack.

> What went wrong or what is missing?
The process of creating the cloudformation stack doesn't complete successfully and this error is shown to me: `No Solution Stack named '64bit Amazon Linux 2018.03 v2.10.0 running Go 1.11.4' found. (Service: AWSElasticBeanstalk; Status Code: 400; Error Code: InvalidParameterValue; Request ID: 27a55fac-ef00-4ee2-8c06-fcc024e23836)`

> Do you have Screenshots?
[image]

> What kind of OS/Browser/Gateway are you using? Which version?
OS: Windows 10
Browser: Google Chrome Version 72.0.3626.109

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

> What do your configuration files look like?

> What do your log files look like?

> Can you fix this yourself and submit a pull request?

Maverick10, 2019-02-24:
I managed to solve the problem by editing the TTN CloudFormation template and uploading it.
I opened the [TTN CloudFormation template](https://s3.amazonaws.com/thethingsnetwork/builds/integration-aws/dist/cloudformation.template) and downloaded. Under `Resources` there is `TTNConfigurationTemplate`, I changed its `SolutionStackName `to `64bit Amazon Linux 2018.03 v2.10.1 running Go 1.11.5`

If you run into the same problem, go to AWS Docs to [Elastic Beanstalk Go supported platforms](https://docs.aws.amazon.com/elasticbeanstalk/latest/platforms/platforms-supported.html#platforms-supported.go) and replace the `SolutionStackName` of `TTNConfigurationTemplate` with the one you find in the doc's url provided.

closed 2019-02-24 by Maverick10

johanstokking, 2019-02-25:
@Maverick10 thanks for reporting. An updated version is being built and pushed right now.

10 karma points for someone who knows how to specify some kind of "latest" here.

tvandergeer, 2020-01-16:
Not (yet) a solution for your karma-bounty for a "latest" parameter, but just a short note that the template needs a new update for the SolutionStackName: "64bit Amazon Linux 2018.03 v2.14.1 running Go 1.13.2"

johanstokking, 2020-01-17:
Thanks, mentioning the solution stack name is already very helpful.

CI and CD is currently running

tvandergeer, 2020-01-17:
Great! Thanks for picking this up instantly. Can I view the outcome of the CI/CD somewhere?

johanstokking, 2020-01-18:
It took about 15 minutes, see updated version in the same place: https://s3.amazonaws.com/thethingsnetwork/builds/integration-aws/dist/cloudformation.template