TheThingsArchive/ttn #763: Can copy data

edgale, 2019-06-19:
This is a **bug report/question** for **the website/the console**.

> What do you want to do?
I got a application set up and in there a view sensors. I want to copy the data from a message of a sensor, but can't because the message is to long. I can't mark the data and copy and can't reach the copy button.
> Do you have Screenshots?
[image]
> What kind of OS/Browser/Gateway are you using? Which version?
Google Chrome Version 75.0.3770.90 (64-Bit)

htdvisser, 2019-06-19:
When I single-click on any character in the payload field, the entire field is selected. Have you tried that?

@kschiffer @bafonins I don't think we'll fix this for v2 anymore, but please confirm that in v3 large payloads also render properly.

edgale, 2019-06-21:
Hi @htdvisser , thanks for your reply. I tired clicking on any character but nothing happens. I can't select the data.

htdvisser, 2019-06-21:
Ah sorry, I overlooked that you're using Chrome. It works in my Firefox, but when I try to copy the contents of the bytes field in Chrome, I can only do that with the copy button. There seems to be a `user-select: none` in there somewhere that prevents selection.

htdvisser, 2021-06-03:
This isn't going to get fixed in V2 anymore, and it should work okay in The Things Stack (V3).

closed 2021-06-03 by htdvisser