TheThingsArchive/ttn #764: Export devices as V3 JSON

htdvisser, 2019-06-21:
This is a **feature request** for **ttnctl**.

> What do you want to do?

I want to migrate devices from v2 to v3.

> What steps did you take?

- In v2: `ttnctl devices list`, `ttnctl devices info [id]`
- In v3: `ttn-lw-cli end-devices create ...`

> What went wrong or what is missing?

It's too much work. It would be a lot easier if there was a `ttnctl devices export [id]` that would print V3 JSON so that I can pipe it directly into `ttn-lw-cli end-devices create`.

> Can you fix this yourself and submit a pull request?

Yes

htdvisser, 2019-06-21:
When this is done and working, we may want to add the same to the console.

htdvisser, 2019-11-22:
Work started in https://github.com/TheThingsNetwork/ttn/compare/feature/v3-export

@rvolosatovs please see what other fields we may need to transfer when migrating existing devices from v2 to v3.

johanstokking, 2020-04-20:
@neoaggelos is taking this over

@htdvisser can you let @neoaggelos know if the branch that you mentioned is still relevant and like we want this to be?

htdvisser, 2020-04-20:
This was practically done. Only thing left was for @rvolosatovs to confirm what MAC settings need to be set in the network server.

johanstokking, 2020-04-20:
OK. @neoaggelos please coordinate with @rvolosatovs and file an issue in `lorawan-stack` for documenting migrating devices from V2 to V3 via CLI.

neoaggelos, 2020-04-21:
> OK. @neoaggelos please coordinate with @rvolosatovs and file an issue in `lorawan-stack` for documenting migrating devices from V2 to V3 via CLI.

There is https://github.com/TheThingsNetwork/lorawan-stack/issues/1685

neoaggelos, 2020-04-29:
I'm having trouble telling apart OTAA from ABP devices from v2 with ttnctl. What I've seen is that all devices (both OTAA and ABP) have `ActivationConstraints` equal to `Local`. The console seems to be  telling them apart by looking up the usages for devaddr prefixes in https://console.thethingsnetwork.org/api/handlers.

`ttnctl discover broker --metadata` is the closest I can get, but it does not include the usages of the prefixes.

Can we retrieve this info from the handler itself?

johanstokking, 2020-04-29:
I don't think you can get this from the Handler, but you should be able to get this from the V2 NS.

htdvisser, 2020-04-29:
The v2 API only allows talking to the Handler, so you can't talk to the NetworkServer.

The difference between OTAA and ABP devices in v2 is that OTAA devices have an AppKey and ABP devices don't.

PR #776 (2020-05-03): Export v2 devices in v3 format
## Summary

Closes #764 . This is the v2 part of migrating end devices to the v3.

## Changes

- Adds `ttnctl devices export <device-id>` command
- Adds `ttnctl devices export-all` command

## Example

```
$ ttnctl devices export-all --frequency-plan-id EU_863_870 | ttn-lw-cli dev create --application-id my-v3-app
```

It is required to pass the frequency plan ID (v2 devices don't have one). We could probably pick a default one from the handler/router, but I think it's better to let the user decide.

## Review

@htdvisser please review CLI and usage. Also, runnin [... cut]
closed 2020-05-03 by neoaggelos