TheThingsArchive/ttn #765: ipv6 only http integration target

adrianimboden, 2019-06-28:
Hi

I have a question. We use the http integration to forward the received data to our backend application.
We have many of those (prod, testing, devel, ...). The developement instances are ipv6 only (as we don't have multiple ipv4 addresses, this seemed to be a viable solution).

The problem is, I don't see packages coming into the ipv6 targets while it works for ipv4 targets.

- Is there a possibility to see errors on the ttn console?
- Is ipv6 supported by the HTTP integration?

Thank you very much.

johanstokking, 2019-06-28:
I'm assuming you're referring to V2, not V3, hence the transfer of this issue to the right repository.

In V2 with the current host, we do not support IPv6 unfortunately.

closed 2019-06-28 by johanstokking

adrianimboden, 2019-06-28:
:+1: ok thank you