TheThingsArchive/ttn #766: FSK uplinks does not appear on console

jpmeijers, 2019-07-14:
This is a bug report for the backend.

> What do you want to do?

Uplinks on the FSK channel.

> What steps did you take?

Do an OTAA join. Configure the FSK channel manually as the network does not configure it. Force the device to DR7 (FSK).

> What went wrong or what is missing?

The uplinks are transmitted. I can see them on the air using an rtl-sdr dongle. I can also see my gateway receives the packets. When telling the gateway to forward packets to my own computer I can see them being received. The packets do not appear on the gateway traffic tab on the TTN Console. The packets also do not appear under the data tab of the device on the Console.

> Do you have Screenshots?

No

> What kind of OS/Browser/Gateway are you using? Which version?

I tried both the Semtech UDP packet forwarder as well as the MQTT gateway connector (mp_pkt_fwd).

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

--

> What do your configuration files look like?

--

> What do your log files look like?

When sending the packets to my own computer I can see the FSK packets are received.
```
$ netcat -ul 1700
g��'���؃u{"time":"2019-07-14T14:24:27Z","rxpk":[{"tmst":6945716,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-53,"size":26,"data":"QDskASbAWAABufl0LGVaV5rVMOz3997J1xc="}]}is�'���؃u{"time":"2019-07-14T14:24:34Z","rxpk":[{"tmst":14093845,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-57,"size":26,"data":"QDskASbAWQABk4m1f31lA0RlM5uulHqp5a0="}]}J��'���؃u{"time":"2019-07-14T14:24:41Z","rxpk":[{"tmst":21241966,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-55,"size":26,"data":"QDskASbAWgABkJ4WTGkTpgXzuIVsdcC0XKc="}]})��'���؃u{"time":"2019-07-14T14:24:48Z","rxpk":[{"tmst":28392426,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-57,"size":26,"data":"QDskASbAWwAB+j444xTe9pLk9Cwlplk758Q="}]}���'���؃u{"time":"2019-07-14T14:24:52Z","stat":{"time":"2019-07-14 14:24:52 GMT","lati":-33.93597,"long":18.87081,"alti":0,"rxnb":5,"rxok":4,"rxfw":4,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"IMST + Rpi","mail":"","desc":"JPM <Pi Supply shield>"}}�F�'���؃u{"time":"2019-07-14T14:24:55Z","rxpk":[{"tmst":35540557,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-59,"size":26,"data":"QDskASbAXAABnIbE6m45vyvu4U/o2lXvdko="}]}|��'���؃u{"time":"2019-07-14T14:25:02Z","rxpk":[{"tmst":42688689,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-57,"size":26,"data":"QDskASbAXQABg2D1BnuNCj/APLxc1WSg6U0="}]}T��'���؃u{"time":"2019-07-14T14:25:05Z","rxpk":[{"tmst":45424291,"chan":4,"rfch":0,"freq":867.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.8,"rssi":-72,"size":23,"data":"QIUYASYAnQACV/orE3Q3sUPy7bOftas="}]}��'���؃u{"time":"2019-07-14T14:25:09Z","rxpk":[{"tmst":49836819,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-55,"size":26,"data":"QDskASbAXgABR8uVhm5eDDvkFnbRLDN3YYo="}]}���'���؃u{"time":"2019-07-14T14:25:17Z","rxpk":[{"tmst":56984945,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-55,"size":26,"data":"QDskASbAXwABfse03WjTfJ4tBSZaNGTSTms="}]}.c�'���؃u{"time":"2019-07-14T14:25:22Z","stat":{"time":"2019-07-14 14:25:22 GMT","lati":-33.93597,"long":18.87081,"alti":0,"rxnb":7,"rxok":5,"rxfw":5,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"IMST + Rpi","mail":"","desc":"JPM <Pi Supply shield>"}}3��'���؃u{"time":"2019-07-14T14:25:24Z","rxpk":[{"tmst":64133063,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-54,"size":26,"data":"QDskASbAYAABOGqGMBsuWFqkybMUODvaCmk="}]}
```

> Can you fix this yourself and submit a pull request?

No

htdvisser, 2021-06-03:
This isn't going to get fixed in V2 anymore.

closed 2021-06-03 by htdvisser

kwinz, 2021-08-22:
Just to clarify: Is this now fixed/implemented in V3?