TheThingsArchive/ttn #767: RX2 fallback leads to endless downlinks

sigmaroot, 2019-08-14:
#### Summary
If a MAC command sent in RX2/**SF9** was not answered by an end node, the NS is using RX2/**SF12** as a fallback. In case that the end node IS listening on RX2/**SF9** and just missed the MAC command (due to temporarely unavailability or bad reachability) it will never receive the fallback MACs on RX2/**SF12**. In that case, again no answer is sent by the end node and the NS starts repeating the requests (LinkADRReq and RXParamSetupReq). That produces a huge amount of downlinks that will never be received by the end node that is still listening on RX2/**SF9**.

#### Steps to Reproduce
1. Start end node, join by OTAA
2. Wait for MAC command using RX2/**SF9**
3. Skip answer or shade end node in RX2 window
4. Watch MAC command using RX2/**SF12**
5. See endless repeated downlinks in RX2/**SF12**

#### What is already there? What do you see now?
[image]

#### What is missing? What do you want to see?
Not the same downlink again and again.

#### Environment
Not relevant.

#### How do you propose to implement this?
The RX2 fallback should not be used more than 2 times (for example). After 2 missed answeres, the RX2 should be changed back to **SF9** again. After 3 cycles of RX2/**SF9**-RX2/**SF12** changes some notification should be generated ("end node not answering any MAC") and the downlinks should be stopped. A restart of the procedure after an amount of time may be possible.

#### Can you do this yourself and submit a Pull Request?
No.

mpouillot, 2019-09-11:
It is a important bug, i have few clients which lost its devices due to this bug. I think in OTAA, there is no need to change the SF of the RX2 after the join-accept. If the device receives the join-accept then it is configured to receive in SF9 on RX2. 

For ABP, I think it should be better to configure by using the RX1 window....

sigmaroot, 2019-09-13:
I don't know if everything is correct and compilable (my first contribution in GO 😀), but I've implemented the suggestions of @htdvisser. 

Some things to remind: The `activatedAt` field is never reset to **nil**, I don't know if that has to be done if the device is changed over to ABP or if a new device is registered in that case. I couldn't find any source code for that case. Additionally I found a activation constraint ([here](https://github.com/TheThingsNetwork/ttn/blob/9418424d1d8a8b63908181d383e3642fc5ce2306/core/networkserver/activation.go#L70)) that is already identifying if the device was actived via OTAA. Maybe that can be used for the bypass, too.

Thanks so far!

jpmeijers, 2019-09-28:
I'm currently also affected by this. One of my devices joined using OTAA. Then we had the recent gateway outage and the device never received downlinks. The device (ADR) therefore moved all the way to SF12. Using an SDR I can see it is transmitting on SF12 and the downlinks are also sent on SF12, not on SF9 like it should be.

@sigmaroot @htdvisser how can I assist getting this bugfix tested and out on production TTN?

jpmeijers, 2019-09-28:
Looking at the fallback code, the [line that does the fallback magic](https://github.com/TheThingsNetwork/ttn/blob/1425501427da1c1d9a66a27c6014ced068571af5/core/networkserver/uplink_mac.go#L104) is:
`loraSettings.DataRate = "SF12BW125"`

Can't we change that line to:
```
if loraSettings.DataRate == "SF9BW125" {
    loraSettings.DataRate == "SF12BW125"
} else {
    loraSettings.DataRate == "SF9BW125"
}
```

With this change (if I understand the code correctly) we will alternate between the incorrect (SF12) downlink and the correct (SF9) downlink.

jpmeijers, 2019-10-02:
I'm still stuck with one deployment because of this bug.

Steps to reproduce:
* Only one gateway should be in reach of the device
* Make the device join using OTAA and make sure ADR is switched on
* Over night switch off the gateway
* The next day switch the gateway back on
* The gateway will receive the device on SF12 and downlinks will be sent (ADR, ACK)
* Downlinks are ignored/not received by the device

I can't find a workaround to get the device to receive the downlinks again other than to power cycle it to force a re-join. Is there any other way I can force downlinks back to SF9?

sigmaroot, 2019-10-02:
@jpmeijers 
Please have a look into my commit mentioned earlier: edbe65d5a9a5e04c555166a0fa0b5fb4a712c5b7

terrillmoore, 2019-10-10:
I think there are two different discussions here. One is regarding privately hosted work, another is regarding existing apps deployed using TTN. As I don't have a privately-hosted V2 network, changes for private networks are interesting but don't address our issues. @htdvisser Will pushing a change request on the V2 code make it to production, or is TTN V2 basically frozen at this point, or is there some other way to fix it? Those of us with deployed systems need to understand so we can explain to our stakeholders. It would be good to avoid visits to the devices, and better to avoid firmware updates in devices; but it would be great, no matter how we mitigate, to make sure we only visit devices once. 

Thanks!

htdvisser, 2019-10-11:
As you can imagine, we're trying to spend all our time on v3 and only do really critical things on v2. Of course we do welcome contributions, so if you're willing to implement a fix and test it on a private network, I'll make sure it gets merged and deployed to the public community network.

jpmeijers, 2019-10-14:
I see this as a critical issue as it causes some of my devices to be stuck on SF12, sending retries and draining their batteries. It also prevents me from having remote control of these devices.

Because of this I tried the suggestions in this thread, implemented them on a private instance of the stack on my local computer, and tested. PR #768 contains a working fix.

mpouillot, 2019-10-15:
Exactly, It is a critical issue on public ttn V2. We lost a lot of devices in downlink...

Best regards,

Mathieu Pouillot
Ingénieur Responsable Produit
Tél. : +33(0)4 98 01 60 06

nke Watteco
Valgora Bat C
Avenue Alfred Kastler
83160 LA VALETTE DU VAR – France
http://www.nke-watteco.fr
http://www.nke-watteco.com

closed 2019-10-15 by htdvisser

sigmaroot, 2019-10-15:
Thanks for contributing 😄 
I'm going to investigate the functionality of this fix the next couple of days with our test devices.
Restarting our production devices afterwards.

jpmeijers, 2019-10-15:
I just realised this fix will only be applied to devices that join after the fix is applied. Devices that are using a session from a join before today will still suffer from this bug. On my side that means half of my devices (that automatically rejoins periodically) are now working as expected, while the other half (which never performs a re-join) will still not receive downlinks.

avbentem, 2020-03-16:
As ADR-enabled OTAA devices that joined before October 2019, and all ADR-enabled ABP devices are still running into this<sup>†</sup>, **I wonder if it would be worthwhile to fix this for those in V2 too?**

**ONLY IF YES**, then I wonder:

1. Is there any reason why ABP devices were excluded from making the fallback less persistent? 

   @sigmaroot's [suggested](https://github.com/TheThingsNetwork/ttn/issues/767#issue-480706237):

   > The RX2 fallback should not be used more than 2 times (for example). After 2 missed answeres, the RX2 should be changed back to SF9 again. After 3 cycles of RX2/SF9-RX2/SF12 changes some notification should be generated ("end node not answering any MAC") and the downlinks should be stopped. A restart of the procedure after an amount of time may be possible.

   @htdvisser [suggested](https://github.com/TheThingsNetwork/ttn/issues/767#issuecomment-530839477) what was actually implemented:

   > My suggestion would be to add a "activated at" field to the device so that the NS can detect if it's an OTAA device or not. This field can then be set by HandleActivate after which the fallback can be bypassed for devices that have been activated with OTAA.

   I assume this seemed best/cleanest because OTAA devices should never need the fallback.

    @jpmeijers [suggested](https://github.com/TheThingsNetwork/ttn/issues/767#issuecomment-536227337) something that would work for ABP as well (and, in retrospect, would also work for OTAA devices that joined before October 2019):

    > Looking at the fallback code, the [line that does the fallback magic is](https://github.com/TheThingsNetwork/ttn/blob/v2.10.3/core/networkserver/uplink_mac.go#L104): `loraSettings.DataRate = "SF12BW125"`
    >
    > Can't we change that line to:
    >
    > ```go
    > if loraSettings.DataRate == "SF9BW125" {
    >     loraSettings.DataRate = "SF12BW125"
    > } else {
    >     loraSettings.DataRate = "SF9BW125"
    > }
    > ```
    > With this change (if I understand the code correctly) we will alternate between the incorrect (SF12) downlink and the correct (SF9) downlink.

2. So, what if @jpmeijers' suggestion would be added as well? Or, if that would not work, then what if the fallback would only be used for, say, odd or even values of the downlink (or uplink) counter?

----

<sup>†</sup> A recent case [from Slack](https://thethingsnetwork.slack.com/archives/C1D8GQMU5/p1584106406119400) shows an Elsys OTAA device that dropped to SF12 after some outage, and since already missed 39,000+ (still counting) RX2 downlinks on SF12, with an FOpts of `0503D2AD840345FF0001` telling it to use DR4/SF8. And that single user has about 20 such devices installed.

PR #768 (2019-10-15): Do not fall back to SF12 when activated using OTAA
A modified version of @sigmaroot's fix (https://github.com/TheThingsNetwork/ttn/commit/edbe65d5a9a5e04c555166a0fa0b5fb4a712c5b7) was implemented and tested on a private local instance.

MQTT subscribe output of three different tests attached:
* Original version that incorrectly falls to SF12
* JP's idea of switching between SF12 and SF9 - not working
* Sigmaroot's fix with a small modification to get it to compile - tested to use SF9 to do downlinks. ie working fix.

[TTN private downlink with modified sigmaroot fix.txt](https://github.com/The [... cut]
commit 66d3beaf (2019-10-15): Do not fall back to SF12 when activated using OTAA. Fixes #767