TheThingsArchive/ttn #770: Unable to create device thanks to HTTP API

Oliv4945, 2020-01-03:
This is a bug report for the backend

> What do you want to do?

Add a device to an application thanks to HTTP API

> What steps did you take?

Do a `POST` request to `http://eu.thethings.network:8084/applications/my_app_id/devices/my_dev_id` with content

```
{
   "app_id":"my_app_id",
   "description":"*****",
   "dev_id":"my_dev_id",
   "lorawan_device":{
      "activation_constraints":"local",
      "app_eui":"*****",
      "app_id":"my_app_id",
      "app_key":"*****",
      "dev_eui":"*****",
      "dev_id":"my_dev_id",
      "disable_f_cnt_check":false,
      "f_cnt_down":0,
      "f_cnt_up":0,
      "uses32_bit_f_cnt":true
   }
}
```

> What went wrong or what is missing?

Answer is `permission denied`, also `hivemind_dev_1` is **not** my app_id
```
{
   "error":"permission denied: Broker did not set device: permission denied: NetworkServer did not set device: permission denied: No \"devices\" rights to Application \"hivemind_dev_1\"",
   "message":"permission denied: Broker did not set device: permission denied: NetworkServer did not set device: permission denied: No \"devices\" rights to Application \"hivemind_dev_1\"",
   "code":2
}
```

If think that my request is ok as I get the right answer when I manually create the device in the console and do a `GET` request to `http://eu.thethings.network:8084/applications/my_app_id/devices/my_dev_id`. My application key has all rights selected in the web console

htdvisser, 2020-01-10:
It looks like the AppEUI/DevEUI combination is already taken by another device in this NetworkServer. That device seems to be registered in `hivemind_dev_1`, and you indeed don't have permission to update it.

closed 2020-01-13 by johanstokking

Oliv4945, 2020-01-14:
Hi,

What is surprising is that I can register the device with same AppEui/DevEui in the console, but not the API

thezenox, 2020-02-14:
I am facing the same problem (using node.js, but don't matter). Tried generating different AppEuis and DevEuis, all with `permission denied: No "devices" rights to Application "ttn-tools"` -also not mine. register in web console works with same parameters.