TheThingsArchive/ttn #771: Application Manager http proxy is not secured with an SSL certificate

georgevanvliet, 2020-01-19:
This is a **bug report** for **the backend

> What do you want to do?

I want to talk to the application manager api through a **secure** http proxy

> What steps did you take?

The TTN application manager has [two endpoints](https://www.thethingsnetwork.org/docs/applications/manager/):
* gRPC endpoint, which supports TLS and has [**deprecated**](https://github.com/TheThingsNetwork/node-app-sdk#readme) SDKs for it 
* http endpoint, which proxies the gRPC interface, but has **no https** 

> What went wrong or what is missing?

The http proxy does not have an SSL certificate installed. Therefore this proxy is unsecure.

> Do you have Screenshots?

No, but you can try it by curling to http://eu.thethings.network:8084 and change the scheme to https.

> What kind of OS/Browser/Gateway are you using? Which version?

Mac OS10.15.2 with any browser or http client.

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

N/A

> Can you fix this yourself and submit a pull request?

Nope

pvgennip, 2020-01-19:
Same problem here, please provide it with an SSL certificate.

Thanks in advance for securing this HTTP proxy.

htdvisser, 2021-06-03:
This isn't going to happen in V2 anymore. In V3 all APIs use HTTPS.

closed 2021-06-03 by htdvisser