TheThingsArchive/ttn #773: Location information of TTN Indoor Gateway

sandruen, 2020-03-09:
This is a question about the location information of the TTN Indoor Gateway.

> What do you want to do?
I want to see the latitude and longitude of my gateway in the metadata of the sent packets.

> What steps did you take?
I registered the TTN Indoor Gateway in the console, set a location in the gateway, set all three privacy parameters to public.
My gateway is visible on the TTN map and on TTNMapper.

> What went wrong or what is missing?
In the the TTN console of my device I see the uplink packets, that my gateway receives. In the metadata of my gateway there is no latitude and longitude visible.
As someone suggested, I also queried the NOC url with my gateway ID and also there, the location is empty.
[image]

> What kind of OS/Browser/Gateway are you using? Which version?
OS: Mac, Browser: Firefox, Gateway: TTN Indoor Gateway.

> What are the IDs and EUIs of your Device/Gateway? (if applicable)
Gateway ID: eui-58a0cbfffe8011eb

jpmeijers, 2020-03-24:
This is a known issue with Basic Station protocol gateways that connect to the V2 TTN stack. @KrishnaIyer explained this on Slack a couple of months ago.

wizzfizz94, 2021-02-05:
Also interested in finding more information about this issue. Will this be fixed in v3?

htdvisser, 2021-06-03:
This is not going to get fixed in V2 anymore.

It should work when your gateway is connected to V3, but there is still some confusion about how it works exactly. This is tracked in https://github.com/TheThingsNetwork/lorawan-stack/issues/4218.

closed 2021-06-03 by htdvisser