TheThingsArchive/ttn #782: Link ADR loop on US1915

Oliv4945, 2020-07-27:
This is a bug report for the backend

> What do you want to do?

Use TTN V2 with a 8 channels gateway with US915 region. Frequency plan configured as per [this file](https://github.com/TheThingsNetwork/gateway-conf/blob/master/US-global_conf.json)

> What steps did you take?

Start the device, join the network, then wait for the first uplink

> What went wrong or what is missing?

A link ADR loop is created:

The network sends : `JSON down: {"txpk":{"imme":false,"tmst":842666298,"freq":927.5,"rfch":0,"powe":20,"modu":"LORA","datr":"SF7BW500","codr":"4/5","ipol":true,"size":22,"ncrc":true,"data":"YLYkASaKZQADQAIAcQM6AP8BdQ8H4Q=="}}`
This corresponds to mac command contained in the FOpts: `0340020071033A00FF01`; two link ADR commands:
```
0x03 40 02 00 71
=> DataRate_TXPower 0x40
   => DataRate: 0x04
   => TxPower: 0x00
=> Channel Mask 0x00 02
=> Redundancy: 0x71 => 0b0 111 0001
   => ChMaskCntl: 7
      => All 125 kHz OFF. ChMask applies to channels 64 to 71
   => NbTrans: 1

03 3A 00 FF 01
=> DataRate_TXPower 0x3A
   => DataRate: 0x03 (SF7)
   => TxPower: 0x0A => 10
=> Channel mask 0x FF 00
=> Redundancy: 0x 01
   => ChMaskCntl: 0
      => Channels 0 to 15
   => NbTrans: 1
```

So channels 8 to 15 are activated at 125 kHz SF7 + channel 65 at 500 kHz

The modem answers: `JSON up: {"rxpk":[{"tmst":844495452,"chan":4,"rfch":1,"freq":904.700000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.5,"rssi":-69,"size":17,"data":"QLYkASaAaQAA+1QENAfwmHM="}]}` which seems to respect what the LNS asked.
Payload is `03070307` so the device successfully acknowledge both requests.

But the network server creates again the same LinkADRReq command, the modem reply and it never stops. Did I missed something ?

AppSKey: `64441D8DF3E3635710FD46AD16A41E23`
NwkSKey: `9101F5654E83C740859B5F6EC1998161`

Thank you

davideori, 2020-07-28:
To help in debugging @Oliv4945 can you please confirm that this issue is not present in TTN V3?

Thanks

johanstokking, 2020-07-29:
Unfortunately we're not going to fix this issue in V2 anymore. It would require extensive testing to deploy changes at this point, if it would indeed be identified as bug.

closed 2020-07-29 by johanstokking

Oliv4945, 2020-08-31:
Ok, thank you