TheThingsArchive/ttn #789: ttnctl failed to export devices in TTI

cnoork, 2020-11-17:
This is a question for ttnctl.

> What do you want to do?
make an export from devices in TTI

> What steps did you take?
login, make successful list of applications

> What went wrong or what is missing?
get an error when doing an export from a device

> Do you have Screenshots?
yes, as clear text

> What kind of OS/Browser/Gateway are you using? Which version?
Windows 10

> What are the IDs and EUIs of your Device/Gateway? (if applicable)
app-eui: officesense_development
dev-eui: deveui-3135343050376d11

> What do your configuration files look like?
see logging

> What do your log files look like?
see log files

> Can you fix this yourself and submit a pull request?
no
[failed to export 20201117.txt](https://github.com/TheThingsNetwork/ttn/files/5554292/failed.to.export.20201117.txt)

cnoork, 2020-11-18:
When I use the commands in my TTN it is working as long there is only one AppEUI defined, when there are multiple AppEUI in the application it will fail, looks like the second query the header is not filtered for answer index:

C:\_CapgeminiOneDrive\OneDrive - Capgemini\Documents\SmartOffice\TTI\lorawan-stack-cli_2.11_ttn>ttnctl applications select
  INFO Found 6 applications:

=======

        ID                                      Description
1       connected-office-experience-comfort
2       connected-office-experience-desk        OfficeSense Desk
3       connected-office-experience-presence    Officesense Presence
4       noordeindestraat                        noordeindestraat
5       rak811trackernl                         rak811 Tracker
6       thingsuno1                              Testing with Things Uno

Which one do you want to use?
Enter the number (1 - 6) > 3
  INFO Found 2 EUIs

        EUI
1       70B3D5FFFE0B5001
2       70B3D5FFFE0B5501

Which one do you want to use?
Enter the number (1 - 2) >  FATAL Invalid choice for EUI

cnoork, 2020-11-18:
For TTN I get the expected result with the following command:
    > ttnctl devices export-all --app-id connected-office-experience-presence --app-eui 70B3D5FFFE0B5501 --frequency-plan-id EU_863_870
The only trouble is how to get a list from all AppEUI's in an application through the command line.

But still the need is that it is working for TTI, TTN is only to explore the commands.

neoaggelos, 2020-11-30:
Hi @cnoork,

For migrating from The Things Network Stack V2 (e.g. The Things Network) to The Things Stack (V3), please use [`ttn-lw-migrate`](https://github.com/TheThingsNetwork/lorawan-stack-migrate) instead, as `ttnctl` is not actively maintained. See [instructions](https://github.com/TheThingsNetwork/lorawan-stack-migrate#the-things-network-stack-v2).

In order to work with The Things Stack (V3), you will have to use [`ttn-lw-cli`](https://thethingsstack.io/getting-started/cli/) instead.

cnoork, 2020-12-18:
Hi neoaggelos,

Why is there no clue added for the TTI subscription. I tried several options but they are all failing. please give me an example  from a working command.

With kind regards,
Chris

closed 2021-06-21 by htdvisser