TheThingsArchive/ttn #790: Contribute to a new Integration

josmanperez, 2021-03-03:
Hello!, 

I've been working with microprocessors and IoT for some time now but I haven't tried Lora before. I am very new to this world and I would love to contribute while learning.

I sent an email regarding how can I collaborate in creating a new [integration](https://www.thethingsnetwork.org/docs/applications/integrations.html) and they suggested me to write a **question** over the Github repo, so here I am :) 

My idea is to build an integration to connect TTN to [MongoDB Atlas](https://www.mongodb.com/cloud/atlas), I know that currently there is a `Data Storage` integration but I want to also be able to give the community the option to connect through MongoDB and Atlas directly.

I thought about it because MongoDB as a document database could suit most of the projects based on IoT and also because Atlas is a built-in DBaaS with a Free tier option. 

Do you know how can I start? 

Thank you so much,

jpmeijers, 2021-03-03:
Github isn't the right place for questions like these either. Whoever guided you here was mistaken.

Please ask your question on the Forum:
https://www.thethingsnetwork.org/forum/

Or chat to the other TTN community members over on Slack:
https://thethingsnetwork.slack.com/

josmanperez, 2021-03-03:
Hello @jpmeijers , thank you for your answer and sorry for asking the question in the wrong place. I've sent an email to ask_us_everything@thethingsnetwork.org and they guided me through here. 

I will ask the question in the forum. 

Regards!

closed 2021-03-03 by josmanperez