TheThingsArchive/ttn #793: Error fetching device -> Cant delete device from Cluster V2

elka0711, 2021-09-22:
This is a **bug report/feature request/question/...** for **the website/the backend/ttnctl/the console/the NOC/an integration/...**.

> What do you want to do?

> What steps did you take?

> What went wrong or what is missing?

> Do you have Screenshots?

> What kind of OS/Browser/Gateway are you using? Which version?

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

> What do your configuration files look like?

> What do your log files look like?

> Can you fix this yourself and submit a pull request?

elka0711, 2021-09-22:
[image]

elka0711, 2021-09-22:
[image]

elka0711, 2021-09-22:
So it is not possible to bring the sensor up in Cluster V3. 
I believe it is blocked by Cluster V2

elka0711, 2021-09-22:
[image]

elka0711, 2021-09-22:
Fyi,
I am collaborator, with all rights:

- settings
- delete
- devices
- collaborators

htdvisser, 2021-12-14:
With The Things Network V2 shut down, we're going to archive this repository. This issue will be locked.

closed 2021-12-14 by htdvisser